preesm/preesm #231: Infinite loop application

pablosdr, 2019-09-09:
Hi,

This is not an issue with PREESM, but a beginner's question.

How do I create an infinite loop application? I have two actors, one FIFO between then, and I want them not to be executed one single time, but infinite times (as in the Sobel tutorial).

What is the best way to achieve this?

Thanks and regards,

Pablo

leos313, 2019-09-09:
https://github.com/preesm/preesm/issues/209

antoine-morvan, 2019-09-09:
Please have a look at the description of the code generation task : https://preesm.github.io/docs/workflowtasksref/#code-generation-1

Martayugui, 2019-09-30:
I am implementing an algorithm using loops in PREESM. However, when the number of iterations is set to a greater value, this error is displayed: " SEVERE: Trying to set a non strictly positive time for vertex ...." .

How could I fix this issue?

Thank you in advance,

Marta RR.

ahonorat, 2019-09-30:
@Martayugui your bug seems to be the same as issue #234 , please follow this issue instead.

Martayugui, 2019-09-30:
Thank you so much @ahonorat, I was working with a previous version, now it seems to work properly

MariaGarcia7293, 2019-10-02:
Hi,

Sorry for the inconvenience again, I am implementing an algorithm which mainly consists of a "for loop" with several actors executing. So far,this loop seems to be perfectly working for certain number of iterations.

However, in specific cases (with certain values), when I execute my code "SIGNAL SEGMENTATION IS DISPLAYED". I have tracked this error and the program does not even start. 

So far, the code works properly just for certain number of iterations, I am not sure if there is something missing or I have not taken into something into account.

Could you provide any example of application where complex loops are used in PREESM?

Thank  you in advance,

Marta

kdesnos, 2019-10-03:
Dear Maria,

To the best of my knowledge, "SIGNAL SEGMENTATION IS DISPLAYED" is not an error message that is created by any part of the code generated by Preesm.

A close message may be a "segmentation fault" which may appear if, for some reason, the code you have inside an actor tries to access some data outside the allocated input and output buffers. In such a case, try to use valgrind to identify the part of the code responsible for the error.

In any case, all the examples of applications we have are available on the Preesm-Apps github repository. I'm not sure what you mean by "complex loops" but maybe you can find what you're looking fore there.

Best,

Karol

antoine-morvan, 2019-10-03:
Please create a new issue, this is going off topic.

closed 2019-10-03 by antoine-morvan