preesm/preesm #241: Errors in scenario editor

kdesnos, 2019-09-23:
When opening the "Brussels.scenario" scenario of the [org.preesm.poc.adaptiveruntime](https://github.com/preesm/preesm-apps/tree/master/org.preesm.poc.adaptativeruntime) application of Preesm-apps, with the Preesm scenario editor, errors pop-up when opening the "Timing" tab.

3 identical errors are listed in the error windows:
> An error has occurred. See error log for more details.
> Unsupported result type Vector

Even if this error is due to a bad (or old) file content, a better/clearer error message should appear.

Error appeare on:
* Preesm v. 3.17.0
* Windows 10.

kdesnos, 2019-09-23:
Part of the reason may be that the file content was mostly loss during scenario update of commit [Preesm-apps/f963d](https://github.com/preesm/preesm-apps/commit/f963d6ed43ec33c2e0ed603355d74ba38bd21b6b#diff-7923a92f0aa9b9ba10de3bf77f7f6f25)

antoine-morvan, 2019-09-23:
the input scenario has parameter value override strings "[1]", that is considered as a vector by the expression evaluation engine. Replacing the "[1]" by "1" fixes the issue.

antoine-morvan, 2019-09-23:
Todo: add better error support