preesm/preesm #243: Export the scheduling generated Gantt of PREESM to a .dat printable in a Latex paper

mpelcat, 2019-09-26:
When the Gantt chart is displayed in Preesm, it would be very useful to export a .dat (setting a path in the workflow) that can be used to display the Gantt vectorial in a paper.

kdesnos, 2019-09-26:
This may not be the "cleaner" solution, but creating a new Code Generation Printer doing that might do the trick (and I do not think it would be very complex to do).

ahonorat, 2019-09-26:
A xml can already be exported with all firings start and end time with the workflow task org.ietr.preesm.stats.exporter.StatsExporterTask .
What would be the difference ?

ahonorat, 2019-09-26:
[scriptGantt.zip](https://github.com/preesm/preesm/files/3658508/scriptGantt.zip)
I guess this script solved the issue.

closed 2019-09-26 by ahonorat