preesm/preesm #248: App execution not working anymore with the new version of PREESM

leos313, 2019-09-30:
The same application, after the same workflow execution completed successfully, is not working anymore when using PREESM 3.17 and 3.16.
When using PREESM 3.14 (same workflow) the code is generated with no error. When executing it give segmentation fault. The error seems to be in the main when managing threads.
I am going to try to create an little example to reproduce the error. The question is: nobody as found a similar issue?

antoine-morvan, 2019-10-01:
Please provide a minimal example to reproduce.

leos313, 2019-10-01:
I have found the error. It appears only when using PAPIFY.
With version 3.14 the printed code was:

    > 	configure_papify_actor(&papify_actions_initPoint_0,"","initPoint",0/*CodeSetSize*/,"","",0/*numConfigs*/); // Papify --> configure papification of initPoint_0

and everything was working fine.

With Version 3.16, the correspondent printed code is:

    < 	configure_papify_actor(&papify_actions_initPoint_0,"","infiniteLoop/initPoint",0/*CodeSetSize*/,"","",0/*numConfigs*/); // Papify --> configure papification of initPoint_0

and it causes segmentation fault. The difference is in the argument at the third position. Manually replacing `infiniteLoop/initPoint` with `initPoint` solves the problem. With the new version of PAPIFY I hope the problem desappear.

**Question**:

Does it make sense a kind of *version controll* upon the PAPIFY files used in the folder *src* and *include*? They should be compatible with the printed code.

antoine-morvan, 2019-10-01:
@dmadronal does that ring a bell ?

dmadronal, 2019-10-01:
Hi,

Yes, I modified actor names in PREESM codegen to be able to parse automatically papify-output files from scenario UI. I forgot to update the example project, my fault. I will do 2 things and this issue can be closed:
1.- Update the example project with the new version of PAPIFY
2.- Add an entry in troubleshooting for PAPIFY tutorial saying that if an equivalent issue appears, first check with the version from PAPIFY repository

leos313, 2019-10-01:
New version of PAPIFY solves the problem. I have just re-done all the tests

closed 2019-10-03 by antoine-morvan