preesm/preesm #252: Memory scripts for "special" actors do not remove all the unnecessary memcpy calls

Martayugui, 2019-10-07:
I am working with the "develop" branch since I am using the initialization of delays, furthermore,I am working with memory scripts in order to reduce significantly memory. I downloaded the scripts provided by the tutorial of Preesm, but it seems to work just for some "special" actors (broadcast, fork...).

When I analyzed the code generated, memcpy calls which are unnecessary are not removed during the code generation for the all "special" actors.

Here, you can see an example: 

[image]
[image]

Thank you in advance,

Marta RR.

kdesnos, 2019-10-07:
Dear Marta,

Have the input ports of the actors consuming the data been set to "READ_ONLY"? If not, this may explain why the memcpy are still there since, [as explained in the tutorial](https://preesm.github.io/tutos/advancedmemory/#port-annotations), buffer merging making it possible for several actors to access the same data may only be activated if these actor "promise" not to write in the given data, as specified with the Read_only annotation.

If in your code the consumer actors are already marked as read_only, please tell us so, and we'll have a look into the issue.

Best,

Karol

Martayugui, 2019-10-07:
Thank you for your quickly response,

Sorry for the inconvenience I had set all these properties only for the "special actors", not taking into account the consumer actors. This implementation works properly.

Best regards,

Marta RR.

closed 2019-10-07 by kdesnos