preesm/preesm #273: Can't generate SRDAG if Broadcast Actor output rate is 0.

hmiomand, 2019-12-09:
SRDAG generation fails if a **Broadcast Actor** as an output with a rate of 0.
Provided is a simple **.pi** file with a **Broadcast Actor** with a 0 rate output with which a SRDAG can't be generated.

[BrRateZero.zip](https://github.com/preesm/preesm/files/3940104/BrRateZero.zip)

ahonorat, 2019-12-09:
This bug is fixed, but while investigating it, we discover another one related to rates being 0 : the repetition vector is not always correct in this case. This is because current repetition vector computation uses DFS while it should not. We have to update the computation with code of Spider 2.0.

hmiomand, 2019-12-10:
For the following project, the BRV computation is indeed wrong with rates equal to zero.
Also, the code generation errors since the emetteur acteur has no more output fifo (recepteur2 is not fired).

[dummy_project_BRV.zip](https://github.com/preesm/preesm/files/3944217/dummy_project_BRV.zip)

ahonorat, 2020-01-03:
The repetition vector computation has been updated with this new behavior: rates being zero on both sides are allowed, and ports having a rate equal to 0 on special actors are automatically removed if at least one non zero port remains.
This allows to discard full paths without firings in the graph if these paths are delimited by broadcast/roundbuffers for example.

commit d0bca390 (2019-12-09): Partial fix #273 , needs to be tested with refinements, also BRV computation has to be updated
commit 6e738a96 (2020-01-03): Fix #273 completely: mow BRV computation is similar to Spider V2 and handles SDF paths with null rates on special actors, minor updates (int to long, SWT bug id)
closed 2020-01-03 by ahonorat