preesm/preesm #275: Generating a an upper-level graph containing an out of date subgraph will cause an error.

hmiomand, 2019-12-13:
When generating an upper-level graph (with **Generate .diagram**) from a **.pi** file containing a subgraph that had a modification in its refinement (addition/removal of a data port), the generation will fail with a unhelpful error message (see attached image).

[image]

This also prevent the user from seeing the expected ports layout of the subgraph from the point of view of the upper-level graph.

The way to recover from this is to open the .pi file to see the expected ports layout of the subgraph and edit it consequently or to edit the .pi file to fit the new ports layout of the subgraph.

Steps to reproduce:

1. Create the bottom-level graph with 1 input and 1 output data interface.
2. Create the top-level graph containing the bottom-level graph.
3. Add/remove a port from the bottom-level graph.
4. Regenerate the top-level graph with the **Generate .diagram** option.

An error message will be displayed.
From there, the only way to know the expected ports layout of the subgraph seems to be by opening/editing the **.pi** file.

A more convenient outcome would be to display a warning as is done when an actor is out of sync with its **.h** refinement.

hmiomand, 2019-12-13:
Note that adding a **Config. Input Interface** , then generating the graph, yield the expected behavior, simply displaying a warning on the actor despite the missing data interfaces.
It is then possible to know exactly which ports from the bottom-level graph are out of sync from the top-level point of view.

ahonorat, 2021-06-15:
This shoud be fixed by 5236340ac6dfa207bb0388563d556f05ccc7bc07 
Now default behavior is to print a detailed warning (with faulty port name) and generate the top diagram in any case.
Now not only ports from top hierarchical actors are checked, but also the ones of the subgraphs (previously, extra unused ports in  subgraphs were not checked).

commit 5236340a (2023-01-06): (UI) Fix #275
closed 2021-06-15 by ahonorat