preesm/preesm #285: Copy from a subgraph PiGraph with tracking impossible

dylangageot, 2020-01-31:
PGAN scheduler goal is to clusterize actors inside of a specific subgraph in order to generate a schedule. To do so, it's needed to copy the subgraph to apply a first pass of clustering. The problem is that an exception is thrown when copying the subgraph:
```
SEVERE: Copy did not preserve order on parameters
org.preesm.commons.exceptions.PreesmRuntimeException: Copy did not preserve order on parameters
	at org.preesm.model.pisdf.factory.PiMMUserFactory.copyPiGraphWithHistory(PiMMUserFactory.java:110)
	at org.preesm.algorithm.clustering.scheduler.PGANScheduler.<init>(PGANScheduler.java:76)
	at org.preesm.algorithm.clustering.scheduler.PGANSchedulerTask.execute(PGANSchedulerTask.java:38)
	at org.preesm.workflow.AbstractWorkflowExecutor.executeNode(AbstractWorkflowExecutor.java:393)
	at org.preesm.workflow.AbstractWorkflowExecutor.execute(AbstractWorkflowExecutor.java:252)
	at org.preesm.ui.workflow.launch.WorkflowLaunchConfigurationDelegate.launch(WorkflowLaunchConfigurationDelegate.java:86)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:862)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:720)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1029)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$2.run(DebugUIPlugin.java:1243)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

```

The issue can be reproduced by running the workflow of the attached project in the pganScheduler branch (commit 4538cd).

[subgraph_copy.zip](https://github.com/preesm/preesm/files/4140360/subgraph_copy.zip)

ahonorat, 2020-02-18:
A few integration tests are also failing on my branch (commit 252f775) for the same reason.
It seems that the when copying a subgraph, the vertex path is not equal between original and copy since one hierarchy is removed.

I am wondering if this full copy is really necessary, or if the simple copy tracker is okay when dealing with subgraphs. I will test that in 2 weeks.

ahonorat, 2020-03-05:
@dylangageot could you check if the generated code is now correct for your applications?
branch: fixIssue285

dylangageot, 2020-03-09:
Mapping/schedule and code of clustered applications are still correct.

closed 2020-03-20 by jheulot