preesm/preesm #287: Invalid memory address in Legacy memory allocation

orlyn44, 2020-02-07:
Using the new synthesis workflow task with the legacy memory allocation, produced memory address offset are not correctly computed.

Taking the sobel-morpho application one 1 core with a parameter value of 2 for nbSlice, the codegen produces the following output:
[image]

The size indicated in the comment of the Sobel_0.input and Sobel_1.input buffers is correct but the offset of the address is wrong.
Sobel_1.input starts at 50688 when it should start, at least, at 54208. Obviously, this affect all the other addresses that are based on this one which are: 
Dilation_0/1.input, Erosion_0/1.input and y__y__0.
Also, the size of the reserved buffer for Core0 is probably wrong.

kdesnos, 2020-02-07:
A small but important detail: the issue seems to be coming from the connection of the legacy algorithms with the new synthesis task.
From a discussion with @orlyn44 , the issue did not seem to be there in the legacy workflow.

ahonorat, 2020-02-09:
[BugNewMemAlloc.zip](https://github.com/preesm/preesm/files/4177119/BugNewMemAlloc.zip)

Attached project is an attempt to reproduce the error on an MWE. The error is reproducible only with the original graph. It is not reproducible with a similar graph, named "Top". The only difference between the two graphs is the absence of any parameter in "Top".

orlyn44, 2020-02-10:
Things get even weirder, because even though the previous attached project has the READ_ONLY memory port annotation on the input port of the sobel actor, the code generation produce the proper addresses.
However, removing this annotation from the original sobel morpho application graph seems to solve the issue.

orlyn44, 2020-02-10:
Final conclusion: issue is related to the presence of a memory script on the actor split in the original application. Apparently, the new synthesis does not support scripts really well.

ahonorat, 2020-02-16:
Most likely the problem is related to Codegen2, dedicated to the new Synthesis. NULL pointers are not generated when scripts are merging input/output of an actor, whereas the memory allocation seems correct. Also some related memcpy seem to be not correct.

jheulot, 2021-03-25:
By looking to the buffer allocation more in detail, it looks correct with the split memory script.
I joined slice size computation made by @kdesnos where the 50688 appears. 
[image]
Moreover, generating the same graph with the legacy codegen produce the same allocation. 
So, for me, the code generated is correct, I will then close the issue, feel free to reopen it if you find other bad allocations.

ahonorat, 2021-03-25:
Hello, I have tried again, and I think there is still an error.

The error concerns the application of the memory script on the Split actor, leading to a memcpy if using the new synthesis or to nothing if using the previous workflow (latest codegen).

I have attached a new simpler MWE, with updates for scripts, and for parameters of the memory tasks (now they should be identical). Please compare the already generated files in Code/ folder. You will notice the presence of a memcpy in one but not in the other.

Also, a weird thing is that all port annotations are set to NONE, except for the input of Sobel which is READ_ONLY. However, the memory script is on the Split actor, send data to the Sobel actor.

[BugNewMemAlloc.zip](https://github.com/preesm/preesm/files/6205679/BugNewMemAlloc.zip)

commit fc99d8c5 (2021-11-24): (Codegen) Fix #287 : Support of NullBuffer in codegen2
commit 2c89cdf6 (2021-12-24): (Codegen) Fix #287 : Support of NullBuffer in codegen2