preesm/preesm #289: Data input interface Memory Annotation not persisting

hmiomand, 2020-02-12:
Selecting a data interface by clicking on its name, then changing its associated **Memory Annotation**, won't keep the modification.
The **Memory Annotation** need to be set by selecting the interface by clicking on the colored square for the modification to be kept.

ahonorat, 2020-02-13:
This kind of feature also exists for delays. What it does is, in general, automatically done when you set the memory annotation in the properties view: it immediately stores the view value in the model. It is kind of useless but this feature is added automatically by the GUI library in the menu. It is possible to remove it by modifying the application design.

In any case, memory annotations are probably useless on interface ports, we might also consider to not show the annotation in the properties view. This is easily doable. @kdesnos is is ok?

kdesnos, 2020-02-13:
Yes, removing the annotations on port of hierarchical actors may be a good idea since they're most likely ignored in the workflow anyway.

closed 2020-02-18 by ahonorat

ahonorat, 2020-02-18:
Solution choose: GUI does not display the memory annotation of interfaces, and of ports of hierarchical actors. The model still stores these informations.

commit d95c8e4b (2020-02-18): Fix #289 , Fix #274 , rollback what causes #285 error in integration
commit e1daf114 (2020-04-10): Fix #289 , Fix #274 , rollback what causes #285 error in integration