preesm/preesm #295: No corresponding edge found for sink port

hmiomand, 2020-02-24:
Trying to execute the workflow in the attached project gives the error "**No corresponding edge found for sink port out4_u_3**".

[image]

Adding passthrough actors between **Broadcast_u** and **reshape_u** on all but one fifo allows to bypass the issue.
Data-rates and data-types are set to valid values for graph evaluation.
Memory scripts are **not** used in the workflow.

[dummy_no_edge_for_sink.zip](https://github.com/preesm/preesm/files/4243684/dummy_no_edge_for_sink.zip)

orlyn44, 2020-02-24:
Could not reproduce with latest release binaries nor with the master / develop branches.
I'm closing the issue as it seems to be related to your specific configuration.

orlyn44, 2020-02-24:
apparently there IS a bug. 
Here is a minimal project reproducing the bug. 
It seems to happen when there are multiple edges coming from actor to another. 
There must be one actor in between on at least one of the edges for the bug to happen and still at least 2 edges between the two original actors.
A | -> B ->  | C
   | --------> | 
   | --------> | 
In the given exemple, if actor forward is remove there is no problem.
Also, if another actor forward is put between the broadcast and sink, there is no problem.
[issue295.zip](https://github.com/preesm/preesm/files/4244830/issue295.zip)

commit ba61d822 (2020-02-24): Fix #295
commit 13a13145 (2020-02-25): Merge pull request #298 from preesm/fixIssue295

Fix #295
closed 2020-02-25 by jheulot