preesm/preesm #302: (GUI) Severe: UI error release 3.19

vmorillon, 2020-03-09:
OS: Ubuntu 18
Java version: openjdk 11.0.6 2020-01-14

The following project contains a bug that can be reproduced:
[bug.zip](https://github.com/preesm/preesm/files/4305283/bug.zip)

Once the project unzipped, use the following procedure:
- Open project directory
- Open Algo directory
- Open Archi directory
- Open test.diagram (double-click)
- Open test.pi (double-click) (if it throws a java.lang.NullPointerExeption (~10-20%), please restart Preesm)
- Open 1CoreX86.slam (double-click)
- Open 8CoreX87.slam (double-click)
- Generate test.diagram from test.pi (right-click test.pi -> Preesm -> Generate .diagram)
- Overwrite -> OK
- Discard -> Yes

Most of the times (~80-90%) it throws a java.lang.NullPointerExeption. Sometimes this exception can even freeze the eclipse window (~10-20%). If the .diagram is generated without error restart Preesm.

ahonorat, 2020-03-09:
I think this is because Graphiti has been coded by wizards. Most probably an easter egg ...

kdesnos, 2020-03-09:
Issue reproduced on Windows version.
Opening only the .pi AND .diagram files is sufficient to reproduce the NullPointerException.

More importantly, the error also occurs when the .diagram is a valid diagram. (in the given files it is corrupted). Steps to reproduce are identical : open the two files, try to generate the .diagram. On my side, it fails 100% of the time.

Potential explanation may come from the way eclipse manages its resources, when the two editors, on the same resources are open. 

The .pi graph visualization with (preesm) graphiti does not need to exist, since the diagram editor base on (eclipse) graphiti is supposed to be the graph editor for Preesm. If removing the "pi" visualization solves the issue, I vote for this. (Make sure the slam and workflow editor still work though).

jheulot, 2020-03-19:
Can anyone check if the PR is resolving the issue? (@kdesnos, @ahonorat)
For me it resolves the problem: the diagram editor is not closed and eclipse have problem to overwrite the file then.

ahonorat, 2020-03-19:
I tried to reproduce and your branch and your fix worked: no error, the .diagram is correctly generated.

commit 73de9686 (2020-03-20): Fix #302 : fix closeEditorIfOpen to check all opened editors
commit 5b092af7 (2020-03-20): Merge pull request #306 from preesm/fix302

Fix #302 : fix closeEditorIfOpen to check all opened editors
closed 2020-03-20 by jheulot