preesm/preesm #303: (Gantt) [release] JVM crash when right-clicking

vmorillon, 2020-03-11:
OS: Ubuntu 18
Java version: openjdk 11.0.6 2020-01-14
preesm: release 3.19

Right-clicking anywhere (or waiting the information pop-up after zooming) on the Solution Gantt crash the eclipse JVM and throw the following error message:

[image]

ahonorat, 2020-03-11:
I think this bug appears because Preesm 3.19 uses the version 4.09 of Eclipse.
I hope that modifying dependencies in the Maven build to the upcoming Eclipse 2020-03 will solve the problem.

@jheulot can you change the Eclipse dependency once the new Eclipse release will be out?

ahonorat, 2021-03-26:
@vmorillon confirmed me that commit 4eb3c44 fixes the issue.

commit 4eb3c445 (2021-02-11): (UI) Finally fix Gantt display on Linux, should fix #303
closed 2021-03-26 by ahonorat