preesm/preesm #320: (PiSDF) [release] inconsistancies when modifying .pi: 'node node has unknown kind'

vmorillon, 2020-05-27:
OS: Ubuntu 18
Java version: openjdk 11.0.7 2020-04-14
preesm: release 3.20 (but also present in 3.19)

## Setup

- Start from a clean project
- Create 2 PiSDF (named "top.diagram" and "sub.diagram")
- In "top.diagram" add an actor and define its refinement as "sub.pi" ("sub.diagramm" can be left empty)

## Step to trigger

- rename "sub.pi" to "sub0.pi" (moving it into an other directory will result in the same state)
- if "top.diagram" is opened, closed it
- try to open "top.diagram", it will fail (IMHO, it should not and should render the diagram with a warning/error in the log)
- close the tab resulting of the error

At this point, the diagram cannot be generate from "top.pi" (obviously due to missing file). The diagram can be recovered by editing manually "top.pi" with any text editor and replacing "sub.pi" by "sub0.pi".
However an other way exists: editing the "top.pi".

- open "top.pi"
- open the basic properties of the actor
- edit the refinement value to "Algo/sub0.pi"
- save and close "top.pi"
- try to generate from "top.pi" -> failure with "parsed node node has unknown kind"

Note: diff before and after saving "top.pi" show a lot of new key labels and some data label in graph

vmorillon, 2021-06-10:
OS: Ubuntu 20.04
Java version: 14.0.2 2020-07-14
preesm: 3.21

An other way to trigger the behavior:
- new preesm project
- create new preesm algorithm (pisdf)
- create an actor named "actor" in the .diagram
- save and close .diagram
- open .pi
- move the "actor" node
- save and close the .pi
- open .diagram
        - failure (the diagram is linked to an old version of the PiSDF meta-model)
- generate .diagram from .pi
        - failure (parsed node node has unknown kind)

ahonorat, 2021-06-15:
Bug has been reproduced, needs investigation. This might be related to #91 .

ahonorat, 2021-09-20:
The old SDF editor seems to be responsible of an incorrect update of the .pi file.
As we have a more modern editor for such files, I suggest to get rid of the old editor in this specific case, simply by removing the .pi file extension in the list of supported extensions.
https://github.com/preesm/preesm/blob/3512ca2bbd123b640741cfd4dc1b3fc8a84c101a/plugins/org.preesm.ui.sdf/plugin.xml#L77

However, if backward compatibility is needed, an update of the xml model or the graphml writer is needed in the old editor, in the graphiti/dftools project: https://github.com/preesm/graphiti
Indeed, the .pi is updated without the node "kind" XML attribute. Other elements are added, whereas they seem useless.

ahonorat, 2021-09-24:
As agreed with @kdesnos , .pi file extension has been removed from the extensions handled by the old 'Preesm Network Editor' previously used for .graphml files.
Thus, clicking on a .pi file now opens the textual editor instead of the old graphml editor.

It would be good to also modify directly the graphiti/dftools project as said in the previous comment, so that the XML/graphml writer of graphiti/dftools can support the new schema of .pi files.

commit 4f38db19 (2021-09-24): (UI) fix #320 , Preesm Network Editor is not anymore the defaut .pi file editor