preesm/preesm #322: (PiSDF) allows weirder C++ header extension

vmorillon, 2020-05-29:
An actor refinement allows only ".h" extension for the header (at least in case of C/C++ implementation).
C++ header naming convention varies a lot depending on developers and in some cases ".h" is already used by a C implemention.

Non exhaustive list of C++ header extension:
- ".h"
- ".hh"
- ".H"
- ".hxx"
- ".hpp"
- ".h++"
- ...

The file filtering can be tracked down to this particular line, but no idea if it will break something else.
https://github.com/preesm/preesm/blob/a0cbf660aa00cbbed327a4f04c8d6ffc41811ea7/plugins/org.preesm.ui.pisdf/src/org/preesm/ui/pisdf/features/AddRefinementFeature.java#L79

kdesnos, 2020-05-29:
Hi, 

Allowing other extensions should not create any issue. 

That being said, a C code header (not C++) was expected when writing the header parser, so I can not guarantee it'll work properly with c++ headers files. 

Karol 

[ https://github.com/preesm/preesm/blob/a0cbf660aa00cbbed327a4f04c8d6ffc41811ea7/plugins/org.preesm.ui.pisdf/src/org/preesm/ui/pisdf/features/AddRefinementFeature.java#L79 | https://github.com/preesm/preesm/blob/a0cbf660aa00cbbed327a4f04c8d6ffc41811ea7/plugins/org.preesm.ui.pisdf/src/org/preesm/ui/pisdf/features/AddRefinementFeature.java#L79 ]

ahonorat, 2021-07-13:
In my branch the header parsing is now done by Eclipse CDT, supporting both C and C++.
I changed the code and now the UI accepts the C++ header file extensions listed by @kdesnos in the previous message.

*However*, this is only for the UI: the codegen doest not support yet these new extensions and will most probably generate incomplete list of include files. Currently, only the .h headers will be listed. Thus, I let the issue open.

A minimal support of C++ could be to wrap all the C++ functions into C ones and to compile the C core separately from the C++ lib on which it depends to. Then, another feature is to support the replacement of C++ templated function parameters by the value of corresponding PiSDF parameters.

commit 925425ff (2021-07-13): (UI) Fix #322 : accepts more C/C++ header file extensions, but not yet supported by codegen
commit d5a2e69c (2021-07-13): (UI) Fix #322 : accepts more C/C++ header file extensions, but not yet supported by codegen