preesm/preesm #336: Adding config actor to the graph / java class error

alemehghasemi, 2021-06-17:
Hello everyone,

I have generated simple graph in PREESM with three actors. 
I am trying to add a reconfigurable actor to set a parameter but I received the following error.
It would be great if you can help me to solve.

****SEVERE: Unexpected Exception: java.lang.ClassCastException:class org.preesm.model.pisdf.impl.ConfigOutputPortImpl cannot be cast to class org.preesm.model.pisdf.Parameter (org.preesm.model.pisdf.impl.ConfigOutputPortImpl and org.preesm.model.pisdf.Parameter are in unnamed module of loader org.eclipse.osgi.internal.loader.EquinoxClassLoader @239bc43f)
 Contact Preesm developers if you cannot solve the problem.****

Thank you.
BR,
Alemeh,

alemehghasemi, 2021-06-17:
Hello ahonorat,
Thank you for your reply.
I followed the same steps explained in the tutorial 9.
I added an actor to the graph named it SliceSizeSetter.
By right click on the actor I added input port config and connected it to the parameter. (output port the same)
I set the header file as refinement.
I use PREESM version 3.21.
You can find attached my project.

Best Regards,
Alemeh,
[ForkJoin_MemScript.zip](https://github.com/preesm/preesm/files/6671860/ForkJoin_MemScript.zip)

ahonorat, 2021-06-18:
Is the following graph what you intent to have?
[image]

I did not have errors while generating this graph on the last night build of Preesm. If the attached graph is what you intent to have, then I will try to reproduce on the same version as you. Interfaces (including Conf. Input) have been heavily modified in the UI and slightly in the model since last revision 3.21. It might be already fixed in the current develop branch.

alemehghasemi, 2021-06-21:
Yes Ahonorat, this is exactly the graph I am trying to generate.
So I should clone PREESM from develop branch?
Thank you for your help.

BR,
Alemeh,

ahonorat, 2021-06-23:
Hello,

sorry for the delay, I tried again with your version of Preesm (3.21) and I did not have any problem. So I am a bit surprised.
Could you try to open the attached project and regenerate the .diagram of the only .pi graph? It should be a MWE of your issue.
[TestIssue336.zip](https://github.com/preesm/preesm/files/6701663/TestIssue336.zip)

Thanks,
Alexandre H.

ahonorat, 2021-09-14:
Hello @alemehghasemi , do you still have the issue?
I no answer, I will close the issue since I cannot reproduce it.

ahonorat, 2021-09-27:
No new input, closing the issue.