preesm/preesm #339: [model] Broadcast/Fork to Fork optimization after flattening is erroneous

ahonorat, 2021-07-27:
[BugFlattenOptim.zip](https://github.com/preesm/preesm/files/6882674/BugFlattenOptim.zip)

In the attached project, flattening the top graph generates generates as usual a flat graph and optimized version of it removing useless special actors.
However, the optimized flat graph is not consistent (some production/consumption are not correct) whereas the regular flat graph is consistent (as expected).

This erroneous behavior is related to the Fork optimization removing a fork an integrating its outputs in the Broadcast preceding it. More specifically, the error seems to be caused by the fact that the BRV of the Broadcast is 2 while the BRV of the Fork is 1.

I guess the solution is to check the total production/consumption equality when performing this optimization. I think this should also be checked for Join to Fork optimization, which currently does not check neither the total production/consumption equality nor the equality of fifo number and rates between Join input and Fork output.

@farresti @kdesnos Could you confirm my guess?

kdesnos, 2021-07-27:
Isn't there a problem in the un-optimized flattened graph?
This looks deadlocked to me because of the two tokens consumed by the inserted "explode" actor.

[image]

Because of the outer loop surrounding the hierarchical actor, the insertion of the explode actor "distributing" the tokens to the two instanciations of the hierarchical actor is not working. Because the second "firing" of the hierarchical actor depends on the first one, the whole outer loop should be unrolled for the flattening to be functional.

ahonorat, 2021-07-27:
I agree: "This looks deadlocked to me because of the two tokens consumed by the inserted "explode" actor."

But ... both the SRDAG generated directely from the initial graph (dec.pi) and the SRDAG generated from its flat *un-optimized* version are exactly the same, and correct!
[image]

kdesnos, 2021-07-28:
OK, I thought that the problem of non-schedulable intermediate "flattened" graphs had been fixed when adopting the new flattener. But as before, this intermediate graph may remain non-schedulable until the single-rate transformation is done.

I agree with your solution to check the "total" production/consumption is preserved when performing the optimizations.

ahonorat, 2021-09-22:
I think one part of the problem was coming from the Special Actor (SA) removal optimization.
This optimization checks Fork and Join and then BroadCast (Bc) and RoundBuffer (Rb) special actors to merge two consecutive of them into only one actor (and remove the other).
Concretely:
- Join1 -> Join2 connection is reduced to the single Join2;
- Join1 -> Rb2 connection is reduced to the single Rb2;
- Fork1 -> Fork2 connection is reduced to the single Fork1;
- Bc1-> Fork2 connection is reduced to the single Bc1;
- Join1 -> Fork2 connection is reduced to none of them (both are removed);
- additionally, SA with only one input and one output having the same rates is removed.

To fix the problem, I have restricted the special actor removal optimization with the following rule:
both production and consumption rates of the FIFO connecting the two consecutive special actors must be equal (except for Join1 -> Fork2). Otherwise, the semantics of the application would be modified by this optimization.

In this issue, the optimization was failing on the case Bc1 -> Fork2, removing Fork2 that was executed once while Bc1 (called "duplicate_ret") is executed twice (since they do not have the same FIFO production and consumption rate).

**However**, I believe there is another problem occurring in the flattening transformation in the case of delays with persistence set to "NONE". This needs more investigations, and a good understanding of the following paper:
[Delays and States in Dataflow Models of Computation](https://hal.archives-ouvertes.fr/hal-01850252)

ahonorat, 2021-09-22:
After checking it, flattening transformation seems to correspond to the paper.
However the intermediate flattened graph is still non schedulable although handled correctly by the SRDAG trnasofmation.

> OK, I thought that the problem of non-schedulable intermediate "flattened" graphs had been fixed when adopting the new flattener. But as before, this intermediate graph may remain non-schedulable [...]

I close this issue but will open a new one for this specific problem.

commit ab6af859 (2021-09-22): (PiMM) partially fix #339 -- restrict the special actor removal optimization
closed 2021-09-22 by ahonorat